Palatis/arduino-softpwm #4: Example is not working.

Koepel, 2017-03-13:
Thank you for the arduino-softpwm library, I like the way it works.

I use it for an ATmega8 by adding "#define TIMSK1 TIMSK".

The example was confusing for me, because the external definition "SOFTPWM_DEFINE_EXTERN_OBJECT_WITH_PWM_LEVELS" is also defined, but not needed for the example.

The example in the loop() is not working. My setup() takes about 10 seconds, therefor micros() has a high value when entering the loop(). The for-loop with 'i' stays at '0' and never gets to the next one. A value is added to micros() with "micros() + WAIT", that introduces the rollover problem. Could you rewrite the example and make it much simpler ? For example a ramp up for each output one by one with normal delays.

Palatis, 2017-03-13:
also, try setting `next = micros() + WAIT` by the end of `setup()` may help.

【Palatis】- Good Jabber！ <palatis@gmail.com> 於 2017年3月13日 週一 23:00 寫道：

> no, I intentionally *NOT* using `delay()` to show the async nature of the
> lib.
> if there is a rollover prob, use `micros() - last > WAIT`, that should
> take care of the roll-over.
>
>
> Thank you for the arduino-softpwm library, I like the way it works.
>
> I use it for an ATmega8 by adding "#define TIMSK1 TIMSK".
>
> The example was confusing for me, because the external definition
> "SOFTPWM_DEFINE_EXTERN_OBJECT_WITH_PWM_LEVELS" is also defined, but not
> needed for the example.
>
> The example in the loop() is not working. My setup() takes about 10
> seconds, therefor micros() has a high value when entering the loop(). The
> for-loop with 'i' stays at '0' and never gets to the next one. A value is
> added to micros() with "micros() + WAIT", that introduces the rollover
> problem. Could you rewrite the example and make it much simpler ? For
> example a ramp up for each output one by one with normal delays.
>
>
>
> Victor Tseng
-- 

Victor Tseng

Koepel, 2017-03-13:
I'm using my own sequence and ramps now, and I'm not going to use micros() like that. Sorry. The goal is to show how the library can be used. I don't see any harm in using delay() for a simple example. Your while-loops are also used as a fixed delay and could be replaced by delayMicroseconds(). 
The correct way to use millis() and micros() is:
`if( micros() - previousMicros > interval)`
The substraction with unsigned long prevents any rollover problem.

I have a few motors, and I can run them with a PWM of 1000Hz on a  8MHz ATmega8 with 30 PWM levels. That's great!

One more thing: The library is for 8-bit AVR chips, and they don't have 'double' floating point. The compiler translates that into single precision 'float'.

Palatis, 2017-03-13:
avoiding `delay()` is intended.

bit the example should be reworked, anyway.
just i dont have freetime to work on this, I'll accept PR :-)

Victor Tseng

PR #10 (2017-08-14): Fix the example.
Fixes #4

Before this patch, the second for loop would run indefinitely because
(v >= 0) is always true if v is unsigned. Also make the LED fade out on
the way.
commit 0d71fe9c (2017-08-14): Fix the example.

Fixes #4

Before this patch, the second for loop would run indefinitely because
(v >= 0) is always true if v is unsigned. Also make the LED fade out on
the way.
closed 2017-08-14 by Palatis