Palatis/arduino-softpwm #8: Question: is there any difference when using.

Fireflaker, 2017-05-09:


Fireflaker, 2017-05-09:
What is the approximate usable frequency range for a 16Mhz clock? Does clock cycle divider work when using a soft pwm/ hard pwm pin? Or the 0-255 counter handles all the work adjusting frequency.

Thank you very much for this library.

Palatis, 2017-05-09:
1. it's a combination of pwm channels, pwm levels, and pwm frequency. you want to update the pin as fast as you can (fast enough so that persistent of vision comes into play), but still level enough processing power to run your own code. use the `printInterruptLoad()` method to determine if the current combination of values is suitable for your application.

2. this lib is a softpwm lib, doesn't use any hw pwm support.

closed 2017-05-09 by Palatis

Fireflaker, 2017-05-10:
Thank you very much！