Palatis/arduino-softpwm #9: DISABLE PWM CHANNEL

curlyz, 2017-06-21:
I have used 8A designed for 6 channel PWM+ Servo in 1 pin, unfortunately , . The problem is that it must be dynamic switching mode between PWM and servo for each pin. How would I do that ?

I have try set PWM command to 0 and servo.attach() , not working.
Any idea?
I think we can work around by setting a function to mask out the channel in SoftPWM::update() ?

Palatis, 2017-06-21:
you can't do this. use different pins.

this requires a lot of checking in the ISR, which is against my design
principle for this lib.

Palatis, 2017-06-21:
ahh, you can probably disable timer ISR altogether which sort of *pause*
the whole pwm subsystem.

yes, it disables all softpwm pins.

【Palatis】- Good Jabber！ <palatis@gmail.com> 於 2017年6月21日 週三 22:41 寫道：

> you can't do this. use different pins.
>
> this requires a lot of checking in the ISR, which is against my design
> principle for this lib.
>
>
>> I have used 8A designed for 6 channel PWM+ Servo in 1 pin, unfortunately
>> , . The problem is that it must be dynamic switching mode between PWM and
>> servo for each pin. How would I do that ?
>>
>> I have try set PWM command to 0 and servo.attach() , not working.
>> Any idea?
>> I think we can work around by setting a function to mask out the channel
>> in SoftPWM::update() ?
>>
>>
>
> Victor Tseng
-- 

Victor Tseng

closed 2017-06-21 by Palatis